satnogs/satnogs-rotator-controller #8: DC Motor Firmware

zisi, 2015-08-17:
PID controller

jdvdperez, 2017-02-22:
Hey guys, there is not much activity here. Are we looking to implement all the features/changes included in the original post on Aug 17, 2015?

zisi, 2017-02-22:
Hey! We are going to implement the above comment in new [PCB](https://github.com/satnogs/satnogs-rotator-controller/pull/27). We must open a discussion here or in IRC or in [satnogs community](https://community.satnogs.org/) about new architecture of rotator controller firmware.

jdvdperez, 2017-02-22:
Sure. What type architecture do you have in mind so far? Event driven, State Machine, command based...

I need to get up to speed.

jdvdperez, 2017-02-23:
I just had a quick overview of the DC and stepper motor controller. Looks like fun! Not much done yet I'd say, right?

jdvdperez, 2017-02-23:
Comments on initial engagement conversations attached...
[List_of_requirements.txt](https://github.com/satnogs/satnogs-rotator-controller/files/796381/List_of_requirements.txt)

zisi, 2017-07-28:
Open this issue in new repo.

closed 2017-07-28 by zisi