satnogs/satnogs-rotator-controller #25: Micro-controller selection

zisi, 2016-11-22:
Specifications:
* Easy to solder (like SOIC20), TQFP-32 is easy to solder(??) or a dev board like arduino pro mini, teensy LC, etc
* Easy to program, for example by using UART (i.e. optiboot)
* Low power (? mW)
* Operate inside a box (IP66) that placed outdoor, so the min-max temperatures are -40C to +100C
* The speed (>10MHz) and computational power of a 8-bit micro-controller is enough for 2 PID controllers, UART handler (easycomm 3) 
* 2 ADC channels (>10bit) for current feedback of dc motor drivers
* 4 PWM channels (>8bit) for dc and stepper motor drivers
* 1 digital output for enable/disable motor drivers
* 1 digital input for status flag of motor drivers
* 2 digital input for end-stops
* 1 UART for communication (or flash the micro-controller
* 1 digital output for RS-485 Half-duplex
* 2 I2C buses for encoders
* Timer for software interrupts
* WDT

alexschultze, 2016-11-22:
Out of personal experience I recommend to stick with the key business and buy as much while you can and not design a new uC board. Stick with what is most commercially available. Simplest and most powerful would be Arduino Mega from your requirements or possibly smaller.

zisi, 2017-02-08:
Finally, we select Αrduino pro-mini with Atmega328p.

* Easy to solder, it requires to solder PTH.
* Optiboot to enable WDT
* 5V@16MHz ~100mW
* One UART ( hardware interrupt) for RS-485 or UART to USB communication and programming
* One 10 bit ADC with 2 channels for current monitoring
* Timer 1, 16bit, for motor controller event
* Timer 2 and 3, 8bit, with 2 channels each, for PWM for DC and stepper motors
* Enough I/O's for motors state
* One I2C bus with multiplexer for encoders and other devices like temperature

closed 2017-02-13 by ppapadeas